h4ckst0ck/dbus-solaredge #5: Solaredge with Battery

HJS72, 2022-10-18:
I have a Solaredge SE8K-RWS48BNN4 with a BYD B-Box LV 7Battery.
Is it possible to include the Battery Parameters in Venus?

h4ckst0ck, 2022-10-19:
Hello, according to the specification there is no modbus register available for the battery:
https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf

HJS72, 2022-10-19:
Kann leider nicht mehr sagen, woher ich die Info hatte - ich hab per Modus folgende Daten aus dem Solaredge erhalten:

`_address	name	description	unit	type	len	factor	offset	formula	role	room	poll	wp	cw	isScale
102837	Batterieleistung	Batterieleistung	W	floatsw	2	1	0		value		true	false	false	false
102853	Batterieladung	Batterieladung	%	floatsw	2	1	0		value		true	false	false	false
`

h4ckst0ck, 2022-10-19:
Ok, es gibt das hier als Addon. Das kannte ich noch nicht.
https://www.photovoltaikforum.com/core/attachment/157903-power-control-open-protocol-for-solaredge-inverters-pdf/

Aber ohne Zugriff auf so ein System ist das recht schwierig das zu implementieren und zu testen. Die Frage die ich mir aber stelle ist, wozu man die Batterieparameter im Venus braucht?

closed 2022-11-03 by h4ckst0ck