h4ckst0ck/dbus-solaredge #6: DbusSolarEdge - ERROR - regs.isError: {regs}

Jautze82, 2022-10-26:
Hi,

I get the following error. What can I do

root@raspberrypi2:~# python /data/dbus-solaredge/dbus-solaredge.py
2022-10-26 10:39:51,928 - DbusSolarEdge - INFO - Startup, trying connection to Modbus-Server: ModbusTCP 192.168.1.221:502, UNIT 2
2022-10-26 10:39:51,931 - DbusSolarEdge - INFO - Connected to Modbus Server.
2022-10-26 10:39:51,950 - root - INFO - registered ourselves on D-Bus as com.victronenergy.grid.grid_id00
2022-10-26 10:39:54,957 - DbusSolarEdge - ERROR - regs.isError: {regs}

Thanks and best regards
Marcel

Jautze82, 2022-10-26:
Hi,
thanks for the quick response.
Here ist the output:

```
root@raspberrypi2:~# python /data/dbus-solaredge/dbus-solaredge.py
2022-10-26 11:13:43,978 - DbusSolarEdge - INFO - Startup, trying connection to Modbus-Server: ModbusTCP 192.168.1.221:502, UNIT 2
2022-10-26 11:13:43,982 - DbusSolarEdge - INFO - Connected to Modbus Server.
2022-10-26 11:13:44,007 - root - INFO - registered ourselves on D-Bus as com.victronenergy.grid.grid_id00
2022-10-26 11:13:47,016 - DbusSolarEdge - ERROR - regs.isError: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
```
Is Unit "2" correct or could it be that I have to change to Unit 1?
Do I have to add the inverter per IP first? Or leave anything empty without system settings?

Thanks and best regards
Marcel

h4ckst0ck, 2022-10-26:
SE Standard is Unit 2. Is you SE Inverter Listening on 192.168.1.221:502?

Jautze82, 2022-10-26:
Hi,

yes it does.

Best regards
Marcel

h4ckst0ck, 2022-10-26:
Give it a try with UNIT = 1, what model of the inverter is this?

Jautze82, 2022-10-26:
I think we change to german as it is mutch easier.

Das habe ich auch schon probiert aber es hat sich nichts geändert.
Gleiches Ergebnis.

Also heute morgen hat es schon mal funktioniert. Dann habe ich neu gestartet dann war es wieder weg. Hatte auch den Inverter per IP hinzugefügt und bekam ihn dann nicht mehr weg.
Danach habe ich nochmal eine neue Version installiert und jetzt bekomme ich es nicht mehr hin.

Jautze82, 2022-10-26:
Ich habe auch ein BMS dran. Manchmal wird es erkannt und manchmal nicht.
Nach 10 mal usb dran und ab geht es dann auf einmal

h4ckst0ck, 2022-10-26:
Hi, ok dann Deutsch :)

h4ckst0ck, 2022-10-26:
Was meinst du mit "Inverter per IP hinzugefügt" ?

h4ckst0ck, 2022-10-26:
Es kann nur einen einzigen Modbus Client an dem SolarEdge geben. Wenn das Victron Zeugs sich da schon hin verbindet geht das nicht. Ich habe auch festgestellt, dass das SE Modbus selbst nicht immer sofort funktioniert - im Zweifel einfach mal den SE neu starten und dann direkt neu versuchen und NUR mit diesem Skript und keinem weiteren Modbus Client.

Jautze82, 2022-10-26:
In den Einstellungen des Venus --> Wechselrichter --> Ip-Adresse manuell hinzufügen --> IP Adresse eintragen dann findet er erst den WR.

h4ckst0ck, 2022-10-26:
Ja, das ist eben der Fehler. Dann baut das Venus die Verbindung via Modbus direkt auf und der Port ist dicht. Das Skript importiert die auch die Werte für den Inverter selbst, eben neben dem Grid-Meter deshalb wird diese Verbindung vom Venus direkt nicht benötigt.

Jautze82, 2022-10-26:
Ja es ist das einzigste.
Aber das Skript schreibt doch connected to Modbus Server.
komisch auch das er das BMS nicht anzeigt.
Das large image von Venus braucht man nicht. Oder?

danke und viele Grüße 
Marcel

h4ckst0ck, 2022-10-26:
Hi, nein ich habe auch kein large Image.

h4ckst0ck, 2022-10-26:
Starte einfach mal den SE neu und checke ob die Verbindung geht.

Jautze82, 2022-10-26:
Hallo,

ich habe das Problem gefunden.

Meine ID ist 126.

Danke und viele Grüße
Marcel

h4ckst0ck, 2022-10-26:
Aha, wow :) Kannst du zufällig ein Screenshot der SetApp machen, wenn du schon dabei bist? Da wo man die ID sieht...

Jautze82, 2022-10-26:
Ich habe vorher die Anleitung genommen.
https://www.victronenergy.com/live/venus-os:gx_solaredge

da müsstest du dir auch den Screenshot klauen können.

Leider habe ich keinen Screenshot vom einrichten.

h4ckst0ck, 2022-10-26:
Perfekt, ok. Ich mache einen Hinweis in das README.

closed 2022-11-03 by h4ckst0ck