h4ckst0ck/dbus-solaredge #14: sometimes modbus grid values are incorrect

bauer1980, 2024-04-21:
Hallo ich stehe vor einem sehr seltsamen Problem. Zu 80% der Zeit stimmen alle Werte und das Skript funktioniert nur teilweise stimmen die Gridzählerdaten überhaupt nicht, dadurch stimmen dann natürlich auch die AC Lasten nicht. 
Ich versuche den Bug schon eine Weile lang zu finden. Es kommen tatsächlich über die Modbusanfrage falsche Daten und dadurch wird die Rechnung falsch. Siehe die folgenden zwei Bilder. Bild 1 sind die richtigen Werte, 3min vorher Bild2 mit falschen Werten beim Gridzähler, den AC-Lasten.

Habe versucht die Update Rate von 200 auf 2000 zu ändern ohne Erfolg. Ich kann das Problem nicht reproduzieren.
[image]
[image]

bauer1980, 2024-04-21:
nach stundemlangen debugging glaube ich hat es etwas mit dem scale faktor zu tun. immer wenn der bei mir aufgrund von kleineren Gridwerten von 0,1 auf 0,01 wechselt dann stimmen alle gridmeterzahlen in in der Folge auch der AC-Load nicht mehr. Zusätzlich kommt solange ein skalefaktor von 0,1 ist ein positiver wert aus dem modbus registern, bei skalefaktor 0,01 kommt ein negativer wert heraus.

cicatrix87, 2024-07-06:
Ich glaube das Thema hatten wir auch im Photovoltaikforum oder? Da kam ja raus, dass eine neue Firmware im Solaredge Wechselrichter zu diesen Problemen führt und man nicht über eine bestimmte Version upgraden sollte. Du kannst ja nochmal deinen Kommentar dazu abgeben, am besten mit den problematischen Versionsständen.

verrueckterKeks, 2024-12-05:
Ich habe das besagte Problem auch mit einem SE17K Inverter und Version 4.22.39.

Ich habe mir die vom Inverter über Modbus ausgegebenen Werte mal genauer angeschaut und ich glaube, dass Solaredge da einen Fehler beim Konvertieren der Werte in die Register 40206-40209 mit Type int16 macht.

Folgendes Beispiel habe ich bei mir gesehen (duch eigene Logausgaben im python script):

2024-12-05 09:53:07,696 - DbusSolarEdge - INFO -     Register 40209   Wert 6806    SF 0.100
2024-12-05 09:53:08,723 - DbusSolarEdge - INFO -     Register 40209   Wert 2322    SF 0.010

Analyse:
Auf der Phase ist die Last 680,6 W (Im Register steht der Wert 6806 und der Scale Factor 0.1)
Bis hierhin ist noch alles gut. Im nächsten Update ist der SF aber 0.01
Die Last auf der Phase hat sich nicht wesentlich geändert.
Auf Grund des SF müsste nun der 10-fache Wert, also 68060 im Register stehen. Das ist aber außerhalb des int16 Wertebereichs. 
68060 ist binär 1_0000_1001_1101_1100
Im Register steht 2322, binär 0000_1001_0001_0010, was einfach nur die 16 least significant Bits von ca. 68060 sind.

Meiner Meinung nach ist der Fehler also bei Solaredge beim Schreiben der Werte in die Modbus Register bzw. dürfte in so einer Situation der SF nicht so klein werden, dass die daraus resultierenden Load Werte nicht mehr in int16 passen.

Als Abhilfe habe ich versucht einen Konsistenz-Check ins python script einzubauen, in der Art
if abs(Power_P1 + Power_P2 + Power_P3 - Power_ALL) > 3)):
   # Modbus Message ignorieren
else:
   # Modbus Message auf dbus schreiben

So könnte man eine ganze Menge fehlerhafter Modbus Messages filtern, aber leider kommt Venus anscheinend nicht mit der reduzierten Update Rate zurecht. 

Ich versuche mal dem Solaredge Support das Problem zu erklären...

toemk1, 2024-12-17:
Hi!

Sieht hier ähnlich aus mit Version 4.22.39 auf dem WR. Ich meine aber zu erkennen, dass es nur falsche Werte gibt, wenn PV-Produktion stattfindet und da auch nur bei größeren Werten (schätzungsweise >1kW). 

Hat sich der SE-Support schon gemeldet? Eigentlich sind die ja recht fix...

vg
t.

verrueckterKeks, 2024-12-17:
Ja, der Support hat sich heute mit einer Nachfrage gemeldet. Falls was rauskommt, melde ich mich.

Außerdem hat der Support ohne sich zu melden anscheinend die Firmware downgegraded. Ich habe jetzt die Version 4.20.36 auf dem Inverter.
Das würde auch erklären, warum ich seit Tagen nur noch einen Scale Factor von 1.0 beobachte und das Problem nicht mehr habe.

Aquarider75, 2025-01-10:
Hallo, 
erstmal Respekt vor der Leistung den Bug aufzuklären. Leider funktioniert es ja nun weiter nicht. Bitte, bitte, bitte hier kommentieren, wenn es irgendeine Lösung gibt. Entweder im Script oder durch Firmware upgrades. Downgrade kann ja allenfalls eine Interimslösung sein. VG

toemk1, 2025-01-10:
+1 von mir.

Bei mir hat sich der Support übrigens geweigert, das Downgrade durchzuführen :(

Aquarider75, 2025-01-10:
> Ich habe das besagte Problem auch mit einem SE17K Inverter und Version 4.22.39.
> 
> Ich habe mir die vom Inverter über Modbus ausgegebenen Werte mal genauer angeschaut und ich glaube, dass Solaredge da einen Fehler beim Konvertieren der Werte in die Register [40206-40209](tel:40206-40209) mit Type int16 macht.
> 
> Folgendes Beispiel habe ich bei mir gesehen (duch eigene Logausgaben im python script):
> 
> 2024-12-05 09:53:07,696 - DbusSolarEdge - INFO - Register 40209 Wert 6806 SF 0.100 2024-12-05 09:53:08,723 - DbusSolarEdge - INFO - Register 40209 Wert 2322 SF 0.010
> 
> Analyse: Auf der Phase ist die Last 680,6 W (Im Register steht der Wert 6806 und der Scale Factor 0.1) Bis hierhin ist noch alles gut. Im nächsten Update ist der SF aber 0.01 Die Last auf der Phase hat sich nicht wesentlich geändert. Auf Grund des SF müsste nun der 10-fache Wert, also 68060 im Register stehen. Das ist aber außerhalb des int16 Wertebereichs. 68060 ist binär 1_0000_1001_1101_1100 Im Register steht 2322, binär 0000_1001_0001_0010, was einfach nur die 16 least significant Bits von ca. 68060 sind.
> 
> Meiner Meinung nach ist der Fehler also bei Solaredge beim Schreiben der Werte in die Modbus Register bzw. dürfte in so einer Situation der SF nicht so klein werden, dass die daraus resultierenden Load Werte nicht mehr in int16 passen.
> 
> Als Abhilfe habe ich versucht einen Konsistenz-Check ins python script einzubauen, in der Art if abs(Power_P1 + Power_P2 + Power_P3 - Power_ALL) > 3)): Modbus Message ignorieren else: Modbus Message auf dbus schreiben
> 
> So könnte man eine ganze Menge fehlerhafter Modbus Messages filtern, aber leider kommt Venus anscheinend nicht mit der reduzierten Update Rate zurecht.
> 
> Edit: Habe das Problem dem Solaredge Support gemeldet und warte auf Rückmeldung.

Wäre es eine Möglichkeit, sobald SF=0.01 kommt, einfach immer den letzten Wert aus Dbus, der SF=0.1 hatte, zu schreiben?

verrueckterKeks, 2025-01-13:
Der Solaredge Support hat das Thema mit dem Firmware Downgrade als gelöst betrachtet. 
Das ist leider keine allgemeingültige Lösung, aber mit der obigen Beschreibung des Problems kann man vielleicht auch ein Downgrade durch den Support erreichen.

@Aquarider75 Mit so einer Idee hatte ich auch experimentiert, aber das hat bei mir nicht funktioniert.

Aquarider75, 2025-01-16:
Leider hat das 4.22.44 den gleichen Bug. Ich habe jetzt einen Victron Energy Meter bestellt. Es war eine schöne Lösung,  nur leider geht es jetzt nicht mehr.

toemk1, 2025-01-16:
Ok, sehr schade, danke für die Info. Bei mir ist leider kein Platz mehr im Verteiler. Außerdem scheint es bei mir trotz der Ausreißer im Schnitt für eine Regelung zu reichen.

Vielleucht auch noch interessant: Ich habe die OpenWB als Wallbox, die ebenfalls das SE-Meter nutzt. Hier stimmen die Werte immer! Leider bin ich kein Entwickler und kann nicht im Code nachlesen, was die anders machen. Oder es fällt da nur nicht auf, weil das Intervall größer ist.

toemk1, 2025-02-01:
Meine Lösung sieht jetzt so aus, dass ich neben der Gesamt-AC-Power nicht mehr die realen Werte der 3 Phasen übergebe sondern den Wert aus dem AC-Power-Register durch 3 teile und dann weitergebe. Das sollte für alle Summen/Statistiken trotzdem passen ist bisher auch für die Regelung sehr zuverlässig. Mir ist nämlich aufgefallen, dass die Summe im Power-Register immer stimmt und nur die Werte der 3 Phasen wild springen.

Oder habe ich einen Denkfehler?

Gruß
t.

toemk1, 2025-02-04:
evtl. habe ich für mich einen workaround gefunden, ich berechne AC Power jetzt selber aus der Stromstärke:
`...
L1
"power": {{ (states('sensor.solaredge_m1_ac_current_a') | float(0))*230 }}
...`

Aktuell sieht das ganz gut aus, ich beobachte weiter....

okurz, 2025-03-27:
Ich war vom gleichen Problem betroffen. Ich habe von https://www.solaredge.com/en/support/setapp-inverters-firmware#Released32111 Firmware Version 4.20.36 runtergeladen und nach https://youtu.be/P_ybrwLRDCo Firmware downgrade von 4.22.44 auf 4.20.36 mit USB Stick (NTFS formatiert) durchgeführt. Dazu Sicherung aus, aufgeschraubt, USB Stick eingesteckt, an, warten. Nach ca. 3m gehen LED blau/grün/rot im Wechsel von unten nach oben. Nach weiteren 3m gestoppt und stabil blau+grün+rot. Mit SetApp Handyapp verbunden und version bestätigt. Danach waren Messwerte wieder in Ordnung. Ich kann für diese Schritte natürlich keinem anderen Garantie geben und nur dazu raten lieber den offiziellen Support oder einen Installateur darum zu bitten das durchzuführen.

h4ckst0ck, 2025-05-10:
> Aktuell sieht das ganz gut aus, ich beobachte weiter....
Wie schaut es aus, ist das eine Lösung die für dich funktioniert?

Senfsaft, 2025-07-23:
Hallo, ich glaube ich habe ein ähnliches Problem. Ich habe den Fehler gemacht meinen Wechselrichter auf 4.23.36 zu updaten. Danach lief erst einmal gar nichts mehr. Da sagte ich mir.. Fuck it. Den CerboGX verpasse ich auch ein Update.

Ich habe das Skript jetzt wieder am laufen. Allerdings funktioniert es beim Akku laden und entladen nicht richtig. Hier stimmen wie bei Euch die AC-Load und Grid Werte nicht mehr. Bei mir springt Phase L1 hin und her. 

Ich weiß nicht, ob wir das gleiche Problem haben, ich wäre allerdings sehr an der Lösung von  @toemk1 interessiert ;) 
Vielleicht hilft mir das ja

toemk1, 2025-07-23:
> Ich weiß nicht, ob wir das gleiche Problem haben, ich wäre allerdings sehr an der Lösung von [@toemk1](https://github.com/toemk1) interessiert ;) Vielleicht hilft mir das ja

Hi!

Ich nutze mitlerweile das hier: [https://github.com/mr-manuel/venus-os_dbus-mqtt-grid](https://github.com/mr-manuel/venus-os_dbus-mqtt-grid)

Von homeassistant aus pushe ich dann nur das Folgende:

`data_template:
  payload: |
    {
      "grid": {
        "power": {{ (states('sensor.grid_power') | float(0))*-1 }},
        "energy_forward": {{ (states('sensor.solaredge_m1_imported_kwh') | float(0))}},
        "energy_reverse": {{ (states('sensor.solaredge_m1_exported_kwh') | float(0))}} 
      }
    }
  topic: homeassistant/energy/grid
action: mqtt.publish`

und

`data_template:
  payload: |
    {
      "pv": {
        "power": {{ (states('sensor.solaredge_ac_power') | float(0)) }},
        "energy_forward": {{ (states('sensor.solaredge_ac_energy_kwh') | float(0)) }}
      }
    }
  topic: homeassistant/energy/pv
action: mqtt.publish`

toemk1, 2025-07-23:
@Senfsaft Hiermit frage in den WR lokal ab: [https://github.com/binsentsu/home-assistant-solaredge-modbus](https://github.com/binsentsu/home-assistant-solaredge-modbus). Bei mir nutze ich zusätzlich noch einen modbus-proxy, weil der SE WR nur eine Verbindung zulässt aber wenn du nur per HA darauf zugreifst, gehts auch direkt. Du musst auf dem WR modbus aktivieren, google mal, das ist einfach.

Senfsaft, 2025-07-23:
Das habe ich mir installiert und die Abfrage funktioniert auch. Die Werte decken sich mit meinem Hauszähler, dass ist schon einmal geil! Jetzt muss ich mich nur noch mit dem publishen der mqtt Nachrichten beschäftigen. Ich habe vor ein paar Tagen mit HA angefangen und bin noch maximal überfordert! Aber vielen vielen Dank für deine Hilfe @toemk1 
Die Werte aus dem Wechselrichter habe ich schon einmal.

toemk1, 2025-08-01:
> Aber woher hast Du diese Werte?
> 
>     * sensor.grid_power
> 
>     * sensor.solaredge_m1_imported_kwh
> 
>     * sensor.solaredge_m1_exported_kwh

Kommt alles vom SolarEdge, teilweise habe ich noch was damit rumgerechnet oder geglättet oder so. Die Rohdaten sind die:

sensor.solaredge_m1_ac_power
sensor.solaredge_m1_imported_kwh
sensor.solaredge_m1_exported_kwh

Viele Grüße!
t.

emphase, 2025-09-19:
> Ich war vom gleichen Problem betroffen. Ich habe von https://www.solaredge.com/en/support/setapp-inverters-firmware#Released32111 Firmware Version 4.20.36 runtergeladen und nach https://youtu.be/P_ybrwLRDCo Firmware downgrade von 4.22.44 auf 4.20.36 mit USB Stick (NTFS formatiert) durchgeführt. 

Ich kann auf der Webseite keine Downloadlinks finden. Hast du deine Firmware-Datei noch und könntest du diese bitte bereitstellen?

okurz, 2025-09-19:
> > Ich war vom gleichen Problem betroffen. Ich habe von https://www.solaredge.com/en/support/setapp-inverters-firmware#Released32111 Firmware Version 4.20.36 runtergeladen und nach https://youtu.be/P_ybrwLRDCo Firmware downgrade von 4.22.44 auf 4.20.36 mit USB Stick (NTFS formatiert) durchgeführt.
> 
> [@okurz](https://github.com/okurz) Ich kann auf der Webseite keine Downloadlinks finden. Hast du deine Firmware-Datei noch und könntest du diese bitte bereitstellen?

Die Downloadlinks sind schwer zu finden und schlecht markiert. In "Table 2; Approved devices’ firmware versions" ist die Version 4.20.36 anklickbar, siehe screenshot

[image]

emphase, 2025-09-20:
Dank dir @okurz Ich habe die Datei geladen, die du markiert hast, auf den USB Stick (1x NTFS und 1x FAT32), nach der Youtube Anleitung angesteckt und gestartet... Auch wie du mehrere Minuten gewartet, leider fängt bei mir der Downgradeprozess nicht an (also das nacheinander Aufleuchten der Farben). Es bleibt bei Dauerlicht aller Farben, bis irgendwann der Normalmodus wieder umgeschwenkt wird (nur blau inkl. grün alle paar Sek am Blinken). 
1. Hast du wirklich diese Datei genau geladen, weil sie ja bei Charger steht? 
2. Könntest du bitte nochmals genau beschreiben, welche Schritte du durchgeführt hast?

emphase, 2025-09-26:
Dank dir nochmals! Auch mit dieser Datei wollte mein Wechselrichter kein Update machen, aber vielleicht hilft es ja nun anderen.

Zwischenzeitlich hat der SolarEdge Support mein Ticket bearbeitet und selbst auf die 4.20.36 downgegradet. Seitdem funktioniert alles wieder.

stueble, 2026-10-16:
Hi, I just noticed the same problem with a firmware version 4.21.xxx and asked the support to downgrade to version 4.20.36. Instead, they updated to version 4.24.519 and claim this version fixes the bug. I could not see the flipping values yesterday, can anybody else confirm that version 4.24.519 fixed the bug discussed here?