h4ckst0ck/dbus-solaredge #15: Change the name of the inverter in Venus OS settings

lobodaa, 2024-08-13:
Hello everyone. First I want to thank h4ckst0ck for a very good job. I managed to set everything up myself without any extra effort. And I also want to ask, is it possible to receive data from several inverters with this driver? And another question, can I edit the name of the inverter? I didn't find it in the settings menu. I will be very grateful for your help.
[image]

mad1982d, 2025-03-18:
Hello, 
just having the same problem after installing a second inverter. How did you get 2 inverters into the venus interface? My original inverter with powermeter is attached via dbus-solaredge, which works very well. But now with my second inverter i don´t know how to do?!?

lobodaa, 2025-04-16:
Hello,

I connected the other inverters using this instruction
https://www.victronenergy.com/live/venus-os:gx_solaredge

________________________________
Від: mad1982d ***@***.***>
Надіслано: 18 березня 2025 р. 16:35
Кому: h4ckst0ck/dbus-solaredge ***@***.***>
Копія: lobodaa ***@***.***>; Author ***@***.***>
Тема: Re: [h4ckst0ck/dbus-solaredge] Change the name of the inverter in Venus OS settings (Issue #15)

[mad1982d]mad1982d left a comment (h4ckst0ck/dbus-solaredge#15)<https://github.com/h4ckst0ck/dbus-solaredge/issues/15#issuecomment-2733483177>

Hello,
just having the same problem after installing a second inverter. How did you get 2 inverters into the venus interface? My original inverter with powermeter is attached via dbus-solaredge, which works very well. But now with my second inverter i don´t know how to do?!?

h4ckst0ck, 2025-05-10:
> want to ask, is it possible to receive data from several inverters with this driver? 

You could simply copy the directory from the driver to a new name, then edit the ids in:
```
dbusservice['grid']           = new_service(base, 'grid',           'grid',              0, 0)
dbusservice['pvinverter.pv0'] = new_service(base, 'pvinverter.pv0', 'pvinverter',        0, 20)
dbusservice['adc-temp0']      = new_service(base, 'temperature',    'temp_pvinverter',   0, 26)
dbusservice['digitalinput0']  = new_service(base, 'digitalinput',   'limit_pvinverter',  0, 10)
```

Then change to IP adress for connecting to the other SE Inverter, do the symlinking to start up the second driver etc...

> And another question, can I edit the name of the inverter? I didn't find it in the settings menu.

You can change this line:
```                
self.add_path('/ProductName', _get_string(regs.registers[0:15])+" "+_get_string(regs.registers[16:31]))
# to
self.add_path('/ProductName', 'Your Productname')
```

lobodaa, 2025-05-26:
Thank you very much for your help.

________________________________
Від: h4ckst0ck ***@***.***>
Надіслано: 10 травня 2025 р. 21:48
Кому: h4ckst0ck/dbus-solaredge ***@***.***>
Копія: lobodaa ***@***.***>; Author ***@***.***>
Тема: Re: [h4ckst0ck/dbus-solaredge] Change the name of the inverter in Venus OS settings (Issue #15)

[https://avatars.githubusercontent.com/u/32590005?s=20&v=4]h4ckst0ck left a comment (h4ckst0ck/dbus-solaredge#15)<https://github.com/h4ckst0ck/dbus-solaredge/issues/15#issuecomment-2869095496>

want to ask, is it possible to receive data from several inverters with this driver?

You could simply copy the directory from the driver to a new name, then edit the ids in:

dbusservice['grid']           = new_service(base, 'grid',           'grid',              0, 0)
dbusservice['pvinverter.pv0'] = new_service(base, 'pvinverter.pv0', 'pvinverter',        0, 20)
dbusservice['adc-temp0']      = new_service(base, 'temperature',    'temp_pvinverter',   0, 26)
dbusservice['digitalinput0']  = new_service(base, 'digitalinput',   'limit_pvinverter',  0, 10)

Then change to IP adress for connecting to the other SE Inverter, do the symlinking to start up the second driver etc...

And another question, can I edit the name of the inverter? I didn't find it in the settings menu.

You can change this line:

self.add_path('/ProductName', _get_string(regs.registers[0:15])+" "+_get_string(regs.registers[16:31]))
# to
self.add_path('/ProductName', 'Your Productname')