h4ckst0ck/dbus-solaredge #20: Remove PV Inverter Limiter Active

Kalored, 2025-05-08:
Hello, 

I am running your script for some years now successfully with Solaredke SE10k and a Victron system and are really happy that it works flawless.

What always bothers me in summer is that the "PV Limiter Active" creates an Alarm when my Inverter reaches the maximum output (10kw). This Alarm not just pops up in the UI of the Cerbo and needs to be clicked away, it also seem to create a high system load on the Cerbo and makes it unresponsive.
I checked and I dont have any Grid feed in limit set and dont want to set any. 

Is there a way to deactivate the alarm or remove the Limiter from the Script? 

I tried to uncomment that line from the script:

dbusservice['digitalinput0']  = new_service(base, 'digitalinput',    'limit_pvinverter', 0, 10)

But that makes the script not starting at all. 

Also I tried to deactivate it in the UI, but all fields this adapter creates cant be changed.

Any other idea to that?

bassmaster187, 2025-05-09:
I have the same problem

h4ckst0ck, 2025-05-10:
Also remove / uncomment this:
```
           if ((regs.registers[36] == 5) & (acpower > 100)):
               dbusservice['digitalinput0']['/State'] = 3
               dbusservice['digitalinput0']['/Alarm'] = 2
           else:
               dbusservice['digitalinput0']['/State'] = 2
               dbusservice['digitalinput0']['/Alarm'] = 0
```
And this, like you have done already:
```
dbusservice['digitalinput0'] = new_service(base, 'digitalinput', 'limit_pvinverter', 0, 10)
```

h4ckst0ck, 2025-05-10:
I also disabled this now per default.

Kalored, 2025-05-12:
Thanks for the answer! Works perfect here.